adafruit/Adafruit_VS1053_Library #14: Fails to compile due to conflicting File classes for Arduino Yun w/ Bridge library

nfriedly, 2015-04-30:
Any time I try to compile a sketch that uses both this lib and the Arduio Yun Bridge library, I get a stack of errors (full output below).

I think this is actually an Arduino bug (https://github.com/arduino/Arduino/issues/1942) but I'm hoping that maybe we can put a workaround into this library, at least until the root issue is resolved. 

I'm happy to put in some work, but I'm not sure what the best way to handle this is. Do you think something like https://github.com/freetronics/FTOLED/commit/7f4476ff19544da09904c94cb0b462eed094c5e5 is reasonable? 

	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::write(unsigned char)'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::flush()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::write(unsigned char const*, unsigned int)'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::operator bool()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::seek(unsigned long)'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::position()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::read()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::peek()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::available()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::read(void*, unsigned int)'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::size()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::close()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::name()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::isDirectory()'
	SD/File.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/File.cpp:57: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::openNextFile(unsigned char)'
	SD/SD.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/SD.cpp:567: first defined here
	Bridge/FileIO.cpp.o: In function `File::write(unsigned char)':
	/Applications/Arduino-1.6.3.app/Contents/Java/libraries/Bridge/src/Bridge.h:52: multiple definition of `File::rewindDirectory()'
	SD/SD.cpp.o:/Applications/Arduino-1.6.3.app/Contents/Java/libraries/SD/src/SD.cpp:234: first defined here
	collect2: error: ld returned 1 exit status
	Error compiling.

nfriedly, 2015-05-06:
BTW, I did a bit of work towards this on https://github.com/nfriedly/Adafruit_VS1053_Library/commit/951ed43d9b0e909a6f641863bc0946ed05290d55 but I'm not very happy with the results so far. It can play streaming audio, but it's super-choppy (at least on my crappy internet connection), and I had to comment out anything that read from the SD card to make it compile. So no pull requests yet...

TheNitek, 2018-01-22:
arduino/Arduino#1942 has been fixed by now, so this one is probably obsolete?

nfriedly, 2018-01-22:
I would hope so, although I'm out on a work trip, so it's going to be more than a week before I can test anything. Since this has gone more than 2 years with no activity, though, let's just close it. Either it works or we can open a new ticket.

closed 2018-01-22 by nfriedly