adafruit/Adafruit_VS1053_Library #15: Unable to compile "player_simple" on Arduino Zero board

Altrane, 2015-06-28:
using the latest 1.6.6 Hourly Build IDE, I'm unable to compile this sketch for Arduino Zero board. 
Copile errors hereunder : 

Arduino : 1.6.6 Hourly Build 2015/06/26 03:43 (Windows 7), Carte : "Arduino Zero (Programming Port)"

Les options de compilation ont été modifiées, tout sera recompilé

D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp: In member function 'void Adafruit_VS1053_FilePlayer::feedBuffer()':
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp:249:14: error: 'SREG' was not declared in this scope
   sregsave = SREG;
              ^
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp:250:7: error: 'cli' was not declared in this scope
   cli();
       ^
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp: In constructor 'Adafruit_VS1053::Adafruit_VS1053(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)':
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp:312:14: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
   clkportreg = portOutputRegister(digitalPinToPort(_clk));
              ^
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp:314:15: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
   misoportreg = portInputRegister(digitalPinToPort(_miso));
               ^
D:\Arduino165+\arduino-nightly\libraries\Adafruit_VS1053_Library-master\Adafruit_VS1053.cpp:316:15: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
   mosiportreg = portOutputRegister(digitalPinToPort(_mosi));
               ^
Erreur lors de la compilation.

  Ce rapport contiendrait plus d'informations si l'option
  "Montrer les informations de sortie pendant la compilation"
  était activée dans Fichier > Préférences.

WALLTECH, 2016-08-24:
Exactly the same issue but for the feather m0 with a cortex m0, please fix

driverblock, 2016-08-25:
Fixed

closed 2016-08-25 by driverblock