adafruit/Adafruit_VS1053_Library #20: feather_midi isn't compiling?

acobarecet, 2017-02-16:
- Arduino board:  Arduino Uno

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.3

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
I did follow the first step and solder the closed jumper and all of the other solder joints are intact. However, when I compiled the feather_midi and uploaded it to my Arduino, it gave me these error messages:
`Arduino: 1.6.3 (Mac OS X), Board: "Arduino/Genuino Uno"

feather_midi.ino: In function 'void setup()':
feather_midi.ino:40:13: error: 'delay' was not declared in this scope
feather_midi.ino:42:3: error: 'Serial' was not declared in this scope
feather_midi.ino:46:3: error: 'VS1053_MIDI' was not declared in this scope
feather_midi.ino:48:43: error: 'midiSetChannelBank' was not declared in this scope
feather_midi.ino:50:30: error: 'midiSetChannelVolume' was not declared in this scope
feather_midi.ino:52:42: error: 'midiSetInstrument' was not declared in this scope
feather_midi.ino: In function 'void loop()':
feather_midi.ino:56:8: error: 'uint8_t' was not declared in this scope
feather_midi.ino:56:22: error: 'i' was not declared in this scope
feather_midi.ino:57:25: error: 'midiNoteOn' was not declared in this scope
feather_midi.ino:58:14: error: 'delay' was not declared in this scope
feather_midi.ino:59:26: error: 'midiNoteOff' was not declared in this scope
feather_midi.ino:62:13: error: 'delay' was not declared in this scope
feather_midi.ino: At global scope:
feather_midi.ino:65:24: error: variable or field 'midiSetInstrument' declared void
feather_midi.ino:65:24: error: 'uint8_t' was not declared in this scope
feather_midi.ino:65:38: error: 'uint8_t' was not declared in this scope
feather_midi.ino:77:27: error: variable or field 'midiSetChannelVolume' declared void
feather_midi.ino:77:27: error: 'uint8_t' was not declared in this scope
feather_midi.ino:77:41: error: 'uint8_t' was not declared in this scope
feather_midi.ino:86:25: error: variable or field 'midiSetChannelBank' declared void
feather_midi.ino:86:25: error: 'uint8_t' was not declared in this scope
feather_midi.ino:86:39: error: 'uint8_t' was not declared in this scope
feather_midi.ino:95:17: error: variable or field 'midiNoteOn' declared void
feather_midi.ino:95:17: error: 'uint8_t' was not declared in this scope
feather_midi.ino:95:31: error: 'uint8_t' was not declared in this scope
feather_midi.ino:95:42: error: 'uint8_t' was not declared in this scope
feather_midi.ino:105:18: error: variable or field 'midiNoteOff' declared void
feather_midi.ino:105:18: error: 'uint8_t' was not declared in this scope
feather_midi.ino:105:32: error: 'uint8_t' was not declared in this scope
feather_midi.ino:105:43: error: 'uint8_t' was not declared in this scope
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`

When I compile and upload the player_miditest, it seems to work fine, so I'm a little confused why this one doesn't work.

ladyada, 2017-02-16:
that demo is for use with feathers only

closed 2017-02-16 by ladyada

snaxxus, 2018-09-21:
//pst... pst...

//these flags aren't defined
//Select one of those defines and adapt it to your board. FYI this dontn't work on the FeatherM0, so I copies #define VS1053_MIDI Serial1 out of the #if block and bwooping ensues. 

#if defined(__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined(TEENSYDUINO) || defined(ARDUINO_STM32_FEATHER)
  #define VS1053_MIDI Serial1
#elif defined(ESP32)
  HardwareSerial Serial1(2);
  #define VS1053_MIDI Serial1
#elif defined(ESP8266)
  #define VS1053_MIDI Serial
#endif

ladyada, 2018-09-21:
woops, can you submit a PR? it would be really helpful!

snaxxus, 2018-09-21:
//Windows 7 32 bit on old work laptop
//Running arduino IDE 1.8.6
//In boards manager added Arduino SAMD boards (32-bits ARM Cortex-M0+) By Arduino version 1.6.19
//^^^^^ It was required
//In boards manager added Adafruit SAMD boards by Adafruit version 1.2.3
//^^^^^ of course we need it (compiles normal examples but complains because the driver disconnects and comes up on a different com port. This is not directly the problem)
//Downloaded Adafruit_VS1053_Library-master.zip and installed it with Manage Libraries in Arduino IDE
//would get vvvvvv
//exit status 1
//'VS1053_MIDI' was not declared in this scope
//until I pulled #define VS1053_MIDI Serial1 out of the #if block below
// I suspect the ARDUINO_SAMD_FEATHER_M0 flag is not defined or another flag should be included in the #if block

#include <Adafruit_VS1053.h>

// Solder closed jumper on bottom!

// See http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf Pg 31
#define VS1053_BANK_DEFAULT 0x00
#define VS1053_BANK_DRUMS1 0x78
#define VS1053_BANK_DRUMS2 0x7F
#define VS1053_BANK_MELODY 0x79

// See http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf Pg 32 for more!
#define VS1053_GM1_OCARINA 80

#define MIDI_NOTE_ON  0x90
#define MIDI_NOTE_OFF 0x80
#define MIDI_CHAN_MSG 0xB0
#define MIDI_CHAN_BANK 0x00
#define MIDI_CHAN_VOLUME 0x07
#define MIDI_CHAN_PROGRAM 0xC0

//QnD Fix vvvv***
#define VS1053_MIDI Serial1
//QnD Fix ^^^^***
#if defined(__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined(TEENSYDUINO) || defined(ARDUINO_STM32_FEATHER)
  #define VS1053_MIDI Serial1
#elif defined(ESP32)
  HardwareSerial Serial1(2);
  #define VS1053_MIDI Serial1
#elif defined(ESP8266)
  #define VS1053_MIDI Serial
#endif

void setup() {
  delay(1000);
  
  Serial.begin(115200);

  Serial.println("VS1053 MIDI test");

  VS1053_MIDI.begin(31250); // MIDI uses a 'strange baud rate'
  
  midiSetChannelBank(0, VS1053_BANK_MELODY);

  midiSetChannelVolume(0, 127);

  midiSetInstrument(0, VS1053_GM1_OCARINA);
}

void loop() {  
  for (uint8_t i=60; i<69; i++) {
    midiNoteOn(0, i, 127);
    delay(100);
    midiNoteOff(0, i, 127);
  }
  
  delay(1000);
}

void midiSetInstrument(uint8_t chan, uint8_t inst) {
  if (chan > 15) return;
  inst --; // page 32 has instruments starting with 1 not 0 :(
  if (inst > 127) return;
  
  VS1053_MIDI.write(MIDI_CHAN_PROGRAM | chan);  
  delay(10);
  VS1053_MIDI.write(inst);
  delay(10);
}

void midiSetChannelVolume(uint8_t chan, uint8_t vol) {
  if (chan > 15) return;
  if (vol > 127) return;

  VS1053_MIDI.write(MIDI_CHAN_MSG | chan);
  VS1053_MIDI.write(MIDI_CHAN_VOLUME);
  VS1053_MIDI.write(vol);
}

void midiSetChannelBank(uint8_t chan, uint8_t bank) {
  if (chan > 15) return;
  if (bank > 127) return;
  
  VS1053_MIDI.write(MIDI_CHAN_MSG | chan);
  VS1053_MIDI.write((uint8_t)MIDI_CHAN_BANK);
  VS1053_MIDI.write(bank);
}

void midiNoteOn(uint8_t chan, uint8_t n, uint8_t vel) {
  if (chan > 15) return;
  if (n > 127) return;
  if (vel > 127) return;
  
  VS1053_MIDI.write(MIDI_NOTE_ON | chan);
  VS1053_MIDI.write(n);
  VS1053_MIDI.write(vel);
}

void midiNoteOff(uint8_t chan, uint8_t n, uint8_t vel) {
  if (chan > 15) return;
  if (n > 127) return;
  if (vel > 127) return;
  
  VS1053_MIDI.write(MIDI_NOTE_OFF | chan);
  VS1053_MIDI.write(n);
  VS1053_MIDI.write(vel);
}