adafruit/Adafruit_VS1053_Library #30: No sound .

harshcht, 2017-06-03:
My player worked fine when I used it for the first time. But now it does not give out any sound.
I tried the player_simple example. here are my observations.(I changed the CS pin of SD card to Pin 34 of Arduino mega).

1)player initializes 
2) it lists all files on the SD-card.
3) It stops at playing track-2 for some time (which roughly matches the time of the audio files).    

note : It does not seems to be the freeze issue as in #4 as the player does not stop.

TheNitek, 2018-01-06:
Are your sure track-2 is a mp3 file? (I had some issues with mp4 files)
What happens if you use play different files? Does it always happen with the 2nd track or always with specific files?

kmccaffr, 2019-06-10:
There is also some issues going from playing OGG recordings then playing mp3's or prerecorded OGG files (even with the same sample rate). 
At least I have found that for myself it will initially play whatever files that aren't VS1053 OGG recordings, then after I play one of the recordings anything else will just be quiet buzzing. Not sure if that's your issue but just some possible things.

caternuson, 2022-09-02:
For that sketch, the second track uses interrupt based playback. So seems related to that. Sometimes these issue get fixed as part of the Board Support Package updates. Closing for now due to age of issue. If problem still existing with current updates to everything, can reopen.

closed 2022-09-02 by caternuson