adafruit/Adafruit_VS1053_Library #34: feedBuffer_noLock not adapted to fast processor

briancouchman, 2017-10-22:
Hello dear Adafruit

I purchased a Feather Huzzah 32 with a ESP32 on it, and a Adafruit Music Maker FeatherWing - MP3 OGG WAV MIDI Synth Player (PID: 3357).
I used the V1053 library and it didn't work. In fact only the first bytes of the data is played for a fraction of a second.
Looking into the code in the feedBuffer_noLock I noticed that if readyForData() returns 0 everything stops, and it seems that the data is fed too fast. After 110 times of writing 32 bytes, readyForData() returns 0 . The datasheet of the chip explains that when that happens the code should just wait for the buffer to be ready again and continue.

I have tentatively changed the code to this 

```
void Adafruit_VS1053_FilePlayer::feedBuffer_noLock(void) {
  if ((! playingMusic) // paused or stopped
      || (! currentTrack)
      || (! readyForData())) {
    return; // paused or stopped
  }

  int cnt = 0;
  // Feed the hungry buffer! :)
  while (playingMusic/*readyForData()*/) {
    // Read some audio data from the SD card file
    int bytesread = currentTrack.read(mp3buffer, VS1053_DATABUFFERLEN);

    if (bytesread == 0) {
      // must be at the end of the file, wrap it up!
      playingMusic = false;
      currentTrack.close();
      break;
    }

    while(!readyForData()){
      //wait for the buffer to be ready to receive more data
    }
    playData(mp3buffer, bytesread);
  }
}
```

It keeps reading the data the until playingMusic becomes false.
But it will wait for readyForData() to be 1 before calling playData()

Hopefully it can be an improvement to the library for controllers like ESP32.
I am very new to whole this so if you see a problem in my code or my thinking it's very welcome.

Thanks

By the way Adafruit, you are so awesome !

TheNitek, 2018-01-21:
I think the problem is a different one: The unterrupts don't work well on the ESP32. Under normal circumstances the interrupt would trigger feedBuffer() when needed.
This gives you 2 options:
- Use playFullFile to play the file. It basically does the same as your change does: Play the file in a blocking method call that returns after the full file is played
- Add something like this to your loop:
void loop() {
  if (musicPlayer.playingMusic) {
    musicPlayer.feedBuffer();
  }
}
This will feed the buffer when needed without blocking for the full playback time.

vogonsaudi, 2019-03-25:
Sadly, neither of these methods seem to work for me with the feather esp32 and the music maker. I can get it to output a wave form, and it sees the file on the card, but won't actually make any sound.

kmccaffr, 2019-06-09:
@briancouchman Unfortunately that's no different than calling play full file, as it locks out everything similarly. @TheNitek The feedbuffer method doesn't update fast enough unless you basically have empty code.

kmccaffr, 2019-06-09:
A quicker (and only temporary shoddy fix) way to go about it, just add a very small delay to loop like delay(500) or even less. It may affect other things but I've found it stabilizes a bit more and allows it to catch up the buffer.

Something to play around with is all.

kmccaffr, 2019-07-25:
I retract what I have previously said, I am using a Particle Photon and I got steady playback for playFullFile but it crashes for using startPlayingFile after there is no more audio to play. So my solution doesn't necessarily work.

PoulpePalpitant, 2021-06-17:
Welp, I am encountering a similar issue with the exact same gear, except the playback is somehow 3 times as fast as it should be. Also the program cant even escape the feedbuffer as long as the track is playing, so no interrupt here. Have you found any solutions to this?

kmccaffr, 2021-06-17:
Oh I had to downgrade the OS, the particle photon introduced some weird overzealous ISR properties around July of 2019.

PoulpePalpitant, 2021-06-18:
This fork from eziya did the trick, everything now works like a charm. Ty the internet
 git@github.com:eziya/ESP32_ADAFRUIT_VS1053.git
 
 I found it from this thread which is kinda similar to this one 
 https://github.com/adafruit/Adafruit_VS1053_Library/issues/45

caternuson, 2022-09-02:
This seems ESP32 specific. Closing in favor of this issue:
https://github.com/adafruit/Adafruit_VS1053_Library/issues/45

See comment in that thread:
https://github.com/adafruit/Adafruit_VS1053_Library/issues/45#issuecomment-1235766662

closed 2022-09-02 by caternuson