adafruit/Adafruit_VS1053_Library #35: ESP8266/NodeMCU: SPI clock speed FIX

startupsolutions, 2017-11-23:
Hi,
I was having glitching problems recording OGG audio with ESP8266 (NodeMCU) due to low SPI clock speed.
I've simply fixed it changing the lines (in **Adafruit_VS1053.cpp**) from:
`
#define VS1053_CONTROL_SPI_SETTING  SPISettings(250000,  MSBFIRST, SPI_MODE0)
#define VS1053_DATA_SPI_SETTING     SPISettings(8000000, MSBFIRST, SPI_MODE0)
`
to:
`
#if defined(ESP8266)
    #define VS1053_CONTROL_SPI_SETTING  SPISettings(1000000,  MSBFIRST, SPI_MODE0)
    #define VS1053_DATA_SPI_SETTING     SPISettings(20000000, MSBFIRST, SPI_MODE0)
#else
    #define VS1053_CONTROL_SPI_SETTING  SPISettings(250000,  MSBFIRST, SPI_MODE0)
    #define VS1053_DATA_SPI_SETTING     SPISettings(8000000, MSBFIRST, SPI_MODE0)
#endif
`
setting speed to 1MHz/20MHz. As I can hear now, no glitches anymore!

ladyada, 2017-11-23:
cool, wanna submit a PR? :)

TheNitek, 2018-01-25:
For me this broke MP3 playback

caternuson, 2022-09-02:
Closing due to age and potentially breaking change.

closed 2022-09-02 by caternuson