adafruit/Adafruit_VS1053_Library #41: VS1053b load patch from SDCard

mycroft2k, 2018-02-05:
Is it possible that will be integrated to the lib?

TheNitek, 2018-02-05:
Jo can do it like this:
```
bool patchVS1053() {
  uint16_t i = 0;

  Serial.println(F("Installing patch to VS1053"));

  SdFile file;
  if (!file.open("patches.053", O_READ)) return false;

  uint16_t addr, n, val;

  while (file.read(&addr, 2) && file.read(&n, 2)) {
    i += 2;
    if (n & 0x8000U) {
      n &= 0x7FFF;
      if (!file.read(&val, 2)) {
        file.close();
        return false;
      }
      while (n--) {
        musicPlayer.sciWrite(addr, val);
      }
    } else {
      while (n--) {
        if (!file.read(&val, 2)) {
          file.close();
          return false;
        }
        i++;
        musicPlayer.sciWrite(addr, val);
      }
    }
  }
  file.close();

  Serial.print(F("Number of bytes: ")); Serial.println(i);
}
```

mycroft2k, 2018-02-07:
thx it works great

akoebbe, 2020-07-19:
I'm having trouble with this solution. The first problem was that the following lines errored out...

```c++
  SdFile file;
  if (!file.open("/patches/flac.028", O_READ)) return false;
```
with the following error
```
src/main.cpp:54:45: error: no matching function for call to 'SdFile::open(const char [18], const uint8_t&)'
```
So I changed the code to the following
```c++
  File file = SD.open("/patches/flac.028", O_READ);
  if (!file) return false;
```
At this point the program can compile, but it doesn't seem like the patch is completely loading. It keeps bailing at the commented line below...

```c++
bool patchVS1053() {
  uint16_t i = 0;

  Serial.println(F("Installing patch to VS1053"));

  SdFile file;
  if (!file.open("patches.053", O_READ)) return false;

  uint16_t addr, n, val;

  while (file.read(&addr, 2) && file.read(&n, 2)) {
    i += 2;
    if (n & 0x8000U) {
      n &= 0x7FFF;
      if (!file.read(&val, 2)) {
        file.close();
        return false;
      }
      while (n--) {
        musicPlayer.sciWrite(addr, val);
      }
    } else {
      while (n--) {
        if (!file.read(&val, 2)) {
          file.close();
          return false; // <--- returning here!!!!
        }
        i++;
        musicPlayer.sciWrite(addr, val);
      }
    }
  }
  file.close();

  Serial.print(F("Number of bytes: ")); Serial.println(i); // I'm never getting here
}
```

Just to be sure, I should be using the .plg files from VLSI, correct?

akoebbe, 2020-07-19:
Ok. So I just looked at the actual .plg file from VLSI. I was expecting to see binary, but instead I'm seeing what looks like C code. I'm getting the feeling I'm not loading the right file. What should I be looking for?

TheNitek, 2020-07-19:
https://raw.githubusercontent.com/madsci1016/Sparkfun-MP3-Player-Shield-Arduino-Library/master/plugins/patches.053

akoebbe, 2020-07-19:
Thanks @TheNitek but that seems like a really old patch. VLSI has patches as recent as 2020-04-29 (http://www.vlsi.fi/en/support/software/vs10xxpatches.html) so how do I get them in to the right format?

akoebbe, 2020-07-19:
I just found the perl script (https://github.com/madsci1016/Sparkfun-MP3-Player-Shield-Arduino-Library/blob/master/plugins/vs_plg_to_bin.pl) in that same repo. I'll give it a shot.

TheNitek, 2020-07-19:
Well there is a converter Perl file in the same project: https://github.com/madsci1016/Sparkfun-MP3-Player-Shield-Arduino-Library/blob/master/plugins/vs_plg_to_bin.pl

akoebbe, 2020-07-19:
Jinks! It worked! Thanks.

caternuson, 2022-09-02:
Closing. Seems resolved.

closed 2022-09-02 by caternuson