adafruit/Adafruit_VS1053_Library #43: Sound dropouts after some times

mycroft2k, 2018-02-24:
once after 10 minutes, then the whole again runs 50 minutes without problems
and then it may be the problem occurs several times in succession

i have a recording attached with the error

[Adafruit_VS1053_Library.zip](https://github.com/adafruit/Adafruit_VS1053_Library/files/1755409/Adafruit_VS1053_Library.zip)

TheNitek, 2018-02-25:
Sounds like you are not feeding the buffer fast enough. But without more information on your setup and software that's impossible to debug

mycroft2k, 2018-02-25:
Hardware Arduino Uno 16Mhz
same problem with SD or SdFat Lib tested with SDCards Sandisk 2GB / Transcend Premium 300 16GB /Transcend Premium 400 32GB 

Software minimal have already the problem see attached file (my final software has over 1900 lines)
[player_interrupts_minimaltest.zip](https://github.com/adafruit/Adafruit_VS1053_Library/files/1756644/player_interrupts_minimaltest.zip)

TheNitek, 2018-02-26:
What about the mp3 file? What's the bitrate? Is it maybe VBR? Is it an MP3? (Had some funny effects trying to play a Audio-MP4)

mycroft2k, 2018-02-26:
it makes no difference if VBR/CBR 128Kbit or 320Kbit MP3 files
the same MP3 goes without problems and at the next runtime i have the problems.

mycroft2k, 2018-03-17:
i have tested it on Arduino M0 same problem
I think the problem is with the memory card it read only 32 byte for the VS1053 and the card is to slow on read for 32byte

is it possible to read from the memory card always 1024byte and write it into a buffer(2048) and send it from buffer to VS1053

mycroft2k, 2018-08-12:
i write my own lib with buffer (2x 12kb) on Arduino M0 and now can play wav files with 44.1khz ;-)
no sound drops more

caternuson, 2022-09-02:
Closing. Seems resolved.

closed 2022-09-02 by caternuson