adafruit/Adafruit_VS1053_Library #44: Crashes ESP8266 after a short while

MarcJenningsUK, 2018-04-15:
- Arduino board:  **Adafruit ESP8266 Feather**

- Featherwing: **Adafruit music maker featherwing**  -  https://www.adafruit.com/product/3436

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.5**

Add MP3 files to SD card and insert in reader
Load example sketch feather_player.ino
Upload and run

After a short time playing (random timing) the ESP will crash and reboot.  Output from serial monitor below

Exception (0):
epc1=0x402045a0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdc0:  40106290 00000200 3ffee8d4 40203256  
3ffffdd0:  ffffffff 00000020 00000001 00000002  
3ffffde0:  00101001 3ffe84cc 00e8ac4d 00000022  
3ffffdf0:  3fffc200 40106258 3fffc258 4000050c  
3ffffe00:  4000437d 00000030 00000011 ffffffff  
3ffffe10:  60000200 00000008 070f0000 80000000  
3ffffe20:  20000000 3fff1250 80000000 203fc060  
3ffffe30:  80000000 3fffc6fc 3ffeeb90 3fff1254  
3ffffe40:  00000214 003fc060 60000600 00000030  
3ffffe50:  40102c76 00000000 3ffeeb6c 4020458c  
3ffffe60:  00000134 40102a72 00000000 402045af  
3ffffe70:  401062f4 00080000 00000016 401062de  
3ffffe80:  ffffffff 00000020 00000000 4000050c  
3ffffe90:  00000000 00000000 0000001f 40100f25  
3ffffea0:  4000050c 40106258 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  feefeffe 3ffeeb90 3ffeeb90 0000002e  
3ffffee0:  ffffffff 3fff0680 3ffeeb90 3fffdab0  
3ffffef0:  00000000 400042db 3ffeed54 00000030  
3fffff00:  40004b31 3fff11f4 00000274 003fc000  
3fffff10:  40101b08 003782f7 3ffe9744 00000000  
3fffff20:  4010694d 3fff11f4 3ffe8e70 01cbaa26  
3fffff30:  00000000 003708e5 3ffe95d0 40220f96  
3fffff40:  4022145a 00000008 60000600 4020f9c1  
3fffff50:  00000000 00000000 4022151f 3ffe9684  
3fffff60:  3ffe8e70 003724b8 3ffe8e70 4020101c  
3fffff70:  40213d05 3ffe9684 3ffe8e70 01cb826a  
3fffff80:  40213d4a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8e90 3fffdad0 3ffeed54 40204edf  
3fffffa0:  40000f49 3ffe9980 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

TheNitek, 2018-04-15:
Interrupt based playback doesn't work very well on the ESP8266.
You can try to remove the following line to get at least the first track to play:
musicPlayer.useInterrupt(VS1053_FILEPLAYER_PIN_INT); (line 99)

caternuson, 2022-09-02:
Closing. Hopefully fixed if using latest ESP8266 BSP release.

Tested the feather_player example from this library on a Feather ESP8266 and it ran as expected.

closed 2022-09-02 by caternuson