adafruit/Adafruit_VS1053_Library #52: "Adafruit_VS1053.cpp" problem with "feather_player" exemple

ComputerGraphicClub, 2019-03-24:
Arduino board: Adafruit ESP32 Feather

Arduino IDE version : 1.8.9

Hi, There is a message error when I try to upload the "feather_player" sketch. 

Here is the message error : 

`/Users/romain/Documents/Arduino/libraries/Adafruit_VS1053/Adafruit_VS1053.cpp: In member function 'long unsigned int Adafruit_VS1053_FilePlayer::mp3_ID3Jumper(fs::File)':
/Users/romain/Documents/Arduino/libraries/Adafruit_VS1053/Adafruit_VS1053.cpp:186:25: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
       if (mp3.read(tag,3)) {

                         ^
In file included from /Users/romain/Documents/Arduino/hardware/espressif/esp32/libraries/SD/src/SD.h:17:0,
                 from /Users/romain/Documents/Arduino/libraries/Adafruit_VS1053/Adafruit_VS1053.h:34,
                 from /Users/romain/Documents/Arduino/libraries/Adafruit_VS1053/Adafruit_VS1053.cpp:15:
/Users/romain/Documents/Arduino/hardware/espressif/esp32/libraries/FS/src/FS.h:60:12: note:   initializing argument 1 of 'size_t fs::File::read(uint8_t*, size_t)'
     size_t read(uint8_t* buf, size_t size);
            ^`

Any issue ? 
Thanks, 

Romain

ladyada, 2019-03-25:
hiya, that's unusual because we have 1.8.8 + ESP32 1.0.1 and no issue - can you try 1.8.8 and ESP32 1.0.01?

ComputerGraphicClub, 2019-03-25:
Hi, thank for reply, 

So I retry with :
Arduino IDE v1.8.8
ESP32 Arduino v1.0.1 (install with Arduino IDE Boards Manager)
Adafruit HUZZAH32 ESP32 Feather + Music Maker FeatherWing
And Adafruit VS1053 Library Version 1.7 (install with Arduino IDE Library Manager)

-> Same problem 

[image]

But, When I try with Adafruit VS1053 Library Version 1.5, it's work !
[image]
[image]

Now, message on the console look like it's working, but there is no sound play from the Jack 3.5mm output. 

I read this : "Note that all but the 32u4-based Feather, playing occurs 'in the background' with an interrupt that handles reading data from the SD card and piping it to the MP3 chip"

Is there something more to do with my setup :  HUZZAH32 ESP32 Feather + Music Maker FeatherWing & SD Card 16Giga  ?

Thank, 

Romain

PR #53 (2019-03-25): Esp32 fix
Fix #52 
Fixes compilation error for ESP32, still compiles fine for esp8266, Arduino Mega und Uno
closed 2019-03-25 by ladyada