adafruit/Adafruit_VS1053_Library #55: record_ogg example does not support multiple recordings without reset

till-h, 2019-04-12:
The record_ogg.ino example does not seem to support recording more than one .ogg file without hard-resetting the board in between recordings. The first file (RECORD00.ogg) is recorded fine and plays back in Audacity etc., of subsequent recordings only the first 1024 bytes make it to the SD card.

Steps to reproduce:

- Open record_ogg.ino, change the following to the music maker shield pins:

    // define the pins used
    #define RESET -1    // VS1053 reset pin (output)
    #define CS 7        // VS1053 chip select pin (output)
    #define DCS 6       // VS1053 Data/command select pin (output)
    #define CARDCS 4    // Card chip select pin
    #define DREQ 3      // VS1053 Data request, ideally an Interrupt pin

- Load record_ogg.ino to the Uno
- Press the REC_BUTTON to do multiple recordings
- Inspect recordings on PC

(Music Maker board on Uno, IDE v1.8.9)

kmccaffr, 2019-06-09:
Resetting the board each time between recordings seems like it would be hard on the IC. 

Just reload the plugin after you have closed the sd file, then give a notification sound when it's ready to record (add a delay before notification if you're additionally worried). Doesn't really seem like an issue unless you want instantaneous multiple recordings, then you should be looking at a beefier device.

caternuson, 2022-09-02:
Does seem like an issue with the example sketch. It's only loading the plugin in setup() and then never again in loop().