adafruit/Adafruit_VS1053_Library #67: Error in compilation for Arduino Uno WiFi Rev2

labrache, 2020-02-07:
Hi !
I'm getting an error when compiling for Uno WiFi Rev2 (i've tried ESP8266 ans Uno and it was OK)
Here is the compile error:
```
Adafruit_VS1053.cpp:48:5: error: 'OCR0A' was not declared in this scope
     OCR0A = 0xAF;
     ^~~~~
Adafruit_VS1053.cpp:48:5: note: suggested alternative: 'O_CREAT'
     OCR0A = 0xAF;
     ^~~~~
     O_CREAT
Adafruit_VS1053.cpp:49:5: error: 'TIMSK0' was not declared in this scope
     TIMSK0 |= _BV(OCIE0A);
     ^~~~~~
Adafruit_VS1053.cpp:49:5: note: suggested alternative: 'TIMERA0'
     TIMSK0 |= _BV(OCIE0A);
     ^~~~~~
     TIMERA0
In file included from arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0,
                 from arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\pgmspace.h:90,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/api/String.h:30,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/api/Print.h:24,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/api/Stream.h:25,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/api/Client.h:22,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/api/ArduinoAPI.h:29,
                 from Arduino15\packages\arduino\hardware\megaavr\1.8.5\cores\arduino/Arduino.h:23,
                 from Arduino\libraries\Adafruit_VS1053_Library/Adafruit_VS1053.h:18,
                 from Adafruit_VS1053.cpp:15:
Adafruit_VS1053.cpp:49:19: error: 'OCIE0A' was not declared in this scope
     TIMSK0 |= _BV(OCIE0A);

```
- Arduino board: Arduino Uno WiFi Rev2
- Arduino IDE version:  1.8.10
- List the steps to reproduce the problem below:
  - I'm using player_example from Example directory and compile it

labrache, 2020-02-09:
I test with files in PR and it's now working, thank you !

ladyada, 2020-02-09:
merged it in!

closed 2020-02-09 by ladyada