adafruit/Adafruit_VS1053_Library #70: Compiler error when using a Arduino Nano Every (ATMega4809)

wraith-ethz, 2020-05-18:
I use a Arduino Nano Every with the ATMega4809 micro controller. By using the Example "feather_player", there are following compiler errors in the file 'Adafruit_VS1053.cpp':
 * 'OCR0A' was not declared in this scope
 * 'TIMSK0' was not declared in this scope
 * 'OCIE0A' was not declared in this scope

ladyada, 2020-05-18:
not surprising, you'll have to update this line
https://github.com/adafruit/Adafruit_VS1053_Library/blob/master/Adafruit_VS1053.cpp#L50
to check that it isnt a nano every 

@facchinm whats the recommended #ifdef way of doing that?

facchinm, 2020-05-19:
@ladyada `ARDUINO_ARCH_AVR` is fine to avoid compiling for 4809 (since the official core adds `ARDUINO_ARCH_MEGAAVR` macro). For compatibility with third party cores, `#ifdef OCR0A` should work better than anything else :wink:

wraith-ethz, 2020-05-19:
@facchinm I checked, and I was using the version 1.0.8 from the Arduino library manager. When I use the master branch library, it compiles now :)
I have now two questions;
 1. does the VS1053 library support the Arduino Nano Every or other ATMega AVRs? I will defiantly test it soon, but maybe you know this already?
 2. will you release the current branch as V1.0.9, so that the fix will be in the official Arduino Library?

Thanks in advance

ladyada, 2020-05-19:
hmm not sure why it has not been bumped - will check the process that manages that, it will get released this week for sure

closed 2020-07-13 by wraith-ethz