adafruit/Adafruit_VS1053_Library #78: how to use the applyPatch() function?

jtlechem, 2022-06-14:
- Arduino board:  Arduino Uno, with vs1053 codec breakout (https://www.adafruit.com/product/1381)

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.19

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): looking for how to use applyPatch() function.

How do you use the applyPatch() function?
Is there any example code out there?

I need to use the patch called "vs1053b-patches-pitch.plg" from https://www.vlsi.fi/en/support/software/vs10xxpatches.html

The loadPlugin() function doesn't work for this, because it seems to be for loading SPI boot images, not .plg files.

TheNitek, 2022-06-14:
I implemented it like this: https://github.com/TheNitek/RfidShelf/blob/25d03f8f40259af9c3f79db7839ce8766bed2723/RfidShelf/src/ShelfPlayback.cpp#L329 using the patch from here: https://github.com/madsci1016/Sparkfun-MP3-Player-Shield-Arduino-Library/tree/master/plugins (which also includes a perl file to convert the files from vlsi.fi)

jtlechem, 2022-06-15:
Thank you, I did my best but doesn't seem to work. I copied that _patchVS1053() function into my Arduino script to just call it locally. Had to tweak a couple things like
`sdfat::File32 file = _SD.open("patches.053", sdfat::O_READ);
  if (!file.isOpen()) {
    return false;
  }`
to
`File file = SD.open("/pshift.053", O_READ);
  if (!file) {
    return false;
  }`
Doesn't throw any errors, but it doesn't seem to load. I tried the patches.053 and pshift.053 files.

Any good ways to simply use the original .plg file from https://www.vlsi.fi/en/support/software/vs10xxpatches.html ?

bimac, 2022-07-09:
[Here's how I did it](https://github.com/bimac/Synkino/blob/16aef7ae5f0a5d1a6a682db10610ff6677be7e37/SynkinoLC_Firmware/SynkinoLC/SynkinoLC.ino#L619-L633) - maybe this helps?

dgnuff, 2024-02-08:
18 months later, here's the solution I used.  The patches I used were downloaded from VLSI's web page at  https://www.vlsi.fi/en/support/software/vs10xxpatches.html

I downloaded the vs1053b-patches290.zip file from the "VS1053b Patches w/ FLAC Decoder" section.  When you look inside that ZIP, there are a bunch of `.c` files.  Pick which one you want and extract it to your project.

It'll have a small routine at the top that is #if 0'ed out, ignore that, and instead go to the bottom and insert the following subroutine:

```
void VS1053Patch(Adafruit_VS1053_FilePlayer &musicPlayer)
{
    int i;
    for (i = 0; i < CODE_SIZE; i++)
    {
        musicPlayer.sciWrite(atab[i], dtab[i]);
    }
}
```
It'll probably also work if the type of the parameter is `Adafruit_VS1053`, since it only actually cares about one routine in there, and it's inherited from the base `Adafruit_VS1053` class.

Add a prototype for that function to your main sketch, and call it immediately after you initially `begin()` the player, or any time you do a hard or soft reset on it.

I could probably hack together a script to convert the data in the `.c` file to something that the `applyPatch()` routine would accept, but I can't be bothered.  In the grand scheme of things, the memory saving would be so miniscule by comparison with the ram available on the ESP32-S2 I'm using that it's just not worth my time.