adafruit/Adafruit_VS1053_Library #79: Not really a issue but..

mikeoverbay, 2022-08-22:
How do you send a string to the library to play of the SD.
I can't find to pass a string variable. It would be handing for making playlists using the found MP3 names.
Thanks :)

caternuson, 2022-08-22:
https://github.com/adafruit/Adafruit_VS1053_Library/blob/aedb7bbce6e83acb5e48ce7a93e10d3a5afe3bf0/examples/player_simple/player_simple.ino#L77

mikeoverbay, 2022-08-23:
That passes a const value. I wanna pass a string variable.

mikeoverbay, 2022-08-24:
By being able to pass a string, I can look up and store every MP3 on the SD and play them back using those string paths.
I set up an array to store the MP3 paths but I can't pass what's stored in the array to the player.
I tried building the library but I can figure out what machine this compiles for.

mikeoverbay, 2022-08-24:
Maybe I'm too old to remember how to do it.. lol

mikeoverbay, 2022-08-24:
This works.. Can't believe how rusty I am at C ffs :(
```
        char arr[list[song_count].length() + 1];
        strcpy(arr, list[song_count].c_str());
        musicPlayer.startPlayingFile(arr)
```;

caternuson, 2022-08-24:
OK. Closing. Seems resolved.

closed 2022-08-24 by caternuson

jacgoudsmit, 2022-08-24:
> This works.. Can't believe how rusty I am at C ffs :(
> 
> ```
>         char arr[list[current_song].length() + 1];
>         strcpy(arr, list[current_song].c_str());
>         musicPlayer.startPlayingFile(arr);
> ```

Based on that code, musicPlayer.startPlayingFile(list[current_song].c_str()) should also work.