adafruit/Adafruit_VS1053_Library #88: Please fix sdfat

BiatuAutMiahn, 2023-05-01:
When using with Adafruit's sdfat library, you get errors and it will refuse to compile.

madias123, 2023-06-09:
Meanwhile there is no way to fix all those sdfat problems for all boards:
sdfat ist the most incompatible library in the whole Arduino land, caused by "to do one's own thing", so we have:
old sdfat version by greiman
new sdfat version by greiman
teensy one
ESP8266 one
Adafruit one
RP2040 one (mostly ESP8266 one, but not really...)
...and many more.

BiatuAutMiahn, 2023-06-11:
Why not target greiman's new repo, or adafruit's. And leave sdfat compatibility for different boards upstream?

madias123, 2023-06-12:
> Why not target greiman's new repo, or adafruit's. And leave sdfat compatibility for different boards upstream?
> 
> Currently I use greiman's...haphazardly.

It's a little bit complicated, because I'm converting an old TAG-MP3-Player-project of mine (STM32duino (Rogers Core), SDFat V1(!) and libraries that I modified in the past using SDfat (like VS1503).
Meanwhile, I converted the project step by step as a whole new project, caused by too old libraries used in the original code and too much hacks/tricks and workaround especially for the STM32 (low level).
Meanwhile I can use my major IO-Code (reading/writing SD Card, sorting dirs and files, reading outMP3-tags) with just the simple
`#include <SD.h>`
So I'm fine with that.

BiatuAutMiahn, 2023-06-13:
Gotcha