adafruit/Adafruit_VS1053_Library #89: Does not work on ESP32

jenschr, 2023-05-02:
The VS1053 library does not work with ESP32's.

- Arduino board: Adafruit ESP32-S3 NO_PSRAM
- Adafruit hardware: Breakout https://www.adafruit.com/product/1381
- Arduino IDE version: 2.0.3 & 2.1.0

Steps to reproduce:
- Open the "player_simple.ino" example after installing the library
- Set pins for BREAKOUT_RESET 9, BREAKOUT_CS 10, BREAKOUT_CS 6, CARDCS 4, DREQ 3
- Connect according to the above
- Compile & upload

This will cause a hardware crash on the ESP32, causing it to reboot at a regular interval. Changing from this constructor:

jenschr, 2023-05-02:
It looks like this issue has been raised in issue #45 but it was closed as a "cannot reproduce". Trying this with this product will make it 100% reproducible https://www.adafruit.com/product/5323

caternuson, 2023-05-03:
The issue is with interrupt playback. That currently is not working for ESP32 platforms. In the `player_simple.ino` example, comment out this line:
```cpp
  musicPlayer.useInterrupt(VS1053_FILEPLAYER_PIN_INT);  // DREQ int
```
and then it should work for the non-interrupt playback call to:
```cpp
  musicPlayer.playFullFile("/track001.mp3");
```
The subsequent interrupt based playback call(s) will **not** work however:
```cpp
  musicPlayer.startPlayingFile("/track002.mp3");
```

Do not have the breakout version of this to test. However, using a Feather ESP32-S3 (PID 5323) and commenting out the DREQ setup in `feather_player.ino` works.

jenschr, 2023-05-03:
I can confirm that removing the interrupt does work. As suggested, playFullFile does play the full file. Using startPlayingFile also works if you make sure to fill the buffer in your loop:

```
void loop() {
  musicPlayer.feedBuffer();
  // more stuff here
}
```

This works as long as the loop time is less than 150ms so the buffer stays full. It would be quite easy to add a timer that can just keep the buffer filled. I might make use this approach. I can do a PR on this if it's useful?

TheNitek, 2023-05-04:
I dont think a timer will work, because thise time depends in the bitrate of the file and probably even more factors. DREQ pin is specifically there to tell you when it is time to send more data, so „watching“ it, with or without interrupt, is the best approach here.

jenschr, 2023-05-04:
Spent some more time on this. What is crashing is reading from SD inside an interrupt. If I just set a flag in the interrupt and then check this flag in the loop, it works fine. Here's an example that is using interrupts, but just not to fill the buffer: https://pastebin.com/qwcJVs8E

It is basically the same as calling the feedBuffer routine from the main loop, but it will only do it when required. I would like to get this to work with interrupts, but I'm unsure where to go next. Any hints helpful!

sidenote: I've tried using the SDFat option present in the library, but it looks like that's not been maintained in a while? Setting PREFER_SDFAT_LIBRARY yields compilation errors. Tried using adafruit/SdFat - Adafruit Fork@^1.5.1 as well as 2.2.1. Slightly different output

jenschr, 2023-06-12:
Turns out that it's not hard to get this working by using FreeRtos Functions. Adding the following code to Adafruit_VS1053.cpp will use a FreeRTOS Task to poll the feedBuffer routine.

I can deliver a PR if you think this is a good way to solve it. I can also test it. My current fork of the library allows me to play files directly from SPIFFS or any other files system.

```
static void feedEsp32( void * pvParameters ){
  for(;;){
    myself->feedBuffer();
    delay(10);
  } 
}

boolean Adafruit_VS1053_FilePlayer::useInterrupt(uint8_t type) {
  myself = this; // oy vey

  if (type == VS1053_FILEPLAYER_TIMER0_INT) {
#if defined(ARDUINO_ARCH_AVR)
    OCR0A = 0xAF;
    TIMSK0 |= _BV(OCIE0A);
    return true;
#elif defined(ESP32)
    TaskHandle_t Task1;
    xTaskCreatePinnedToCore(
                  feedEsp32,   /* Task function. */
                  "Task1",     /* name of task. */
                  10000,       /* Stack size of task */
                  NULL,        /* parameter of the task */
                  1,           /* priority of the task */
                  &Task1,      /* Task handle to keep track of created task */
                  0);          /* pin task to core 0 */  
#elif defined(__arm__) && defined(CORE_TEENSY)
```

Broker89, 2024-07-17:
Thanks a lot @jenschr that fixed it for me!

DenverRick, 2026-02-09:
Fix for VS1053 audio gaps/pauses on ESP32 — bypass feedBuffer() and useInterrupt()

Body:

I was getting periodic audio pauses/gaps (~every 10 seconds) playing music MP3s on an ESP32 Feather + Music Maker FeatherWing. After a lot of debugging, I found three compounding issues:

1. useInterrupt() doesn't work on ESP32 (as noted in this issue) — SD card reads inside ISRs aren't supported on this platform.

2. playFullFile() is too slow without working interrupts. It falls back to calling feedBuffer() with a delay(5) between each call — not fast enough to keep the VS1053 FIFO fed.

3. feedBuffer() reads only 32 bytes per SD card transaction (VS1053_DATABUFFERLEN). SD cards have high per-transaction overhead, so this is very inefficient.

The fix: Don't use useInterrupt(). Use startPlayingFile() for codec setup, then feed the VS1053 yourself — reading 512 bytes from the SD card at a time (matching the native sector size) and sending it in 32-byte chunks with DREQ checks:

#define PLAY_BUF_SIZE 512

void playFile(const char* filename) {
  if (!musicPlayer.startPlayingFile(filename)) return;

  uint8_t buf[PLAY_BUF_SIZE];

  while (musicPlayer.playingMusic) {
    int bytesRead = musicPlayer.currentTrack.read(buf, PLAY_BUF_SIZE);
    if (bytesRead == 0) {
      musicPlayer.playingMusic = false;
      musicPlayer.currentTrack.close();
      break;
    }

    int offset = 0;
    while (offset < bytesRead) {
      while (!musicPlayer.readyForData());
      int chunk = bytesRead - offset;
      if (chunk > 32) chunk = 32;
      musicPlayer.playData(buf + offset, chunk);
      offset += chunk;
    }
  }
}

This reduces SD card transactions by 16x and eliminates the delay(5). Smooth playback on ESP32, no gaps.

Hardware: Adafruit ESP32 Feather + Music Maker FeatherWing (VS1053), SD card with MP3 files.